kigt-inc/charge-cloud-backend-2023 #1: Set up EV Charger change of status webhook interface

kigt-inc, 2023-08-07:
receives EV Charger change of state (change of statuses). See documentation for different status types and what they mean.

kigt-inc, 2023-08-07:
[kigt_event_capture.php.zip](https://github.com/kigt-inc/charge-cloud-backend-2023/files/12271611/kigt_event_capture.php.zip)

kigt-inc, 2023-08-08:
@darshanP-crest @manoj-k-crest 

Note: the CSMS url that is referenced in the webhook code is not correct, that was a prototype built with html, css, js, php

We will be transitioning over to a React client

darshanP-crest, 2023-08-09:
Waiting for Brandon response to what is the trigger point to this webhook and information about it's body.

manoj-k-crest, 2023-08-10:
@kdiaz489 

Here is the video demonstration of webhook testing on Postman and the corresponding database data entry.

https://www.awesomescreenshot.com/video/19839612?key=d09f0213484b510450305b05f9eebb54