kigt-inc/charge-cloud-backend-2023 #2: Transformation of EV Charger timestamps into EV Charger Transactions

kigt-inc, 2023-08-07:
We need to transform timestamps from our database into EV Charger Transactions so to be able to report on the following: 
a.	Charger availability / Charger unavailable 
b.	Charger occupied Transaction 
c.	Charging Event Transaction 
d.	Billing Event 
e.	Demand Response Event

darshanP-crest, 2023-08-09:
Question from our side:
 
What is the flow to convert EV Charger timestamp to EV Charger Station transactions? (Through the calculations?) Is there any formulas or documents for this?

Waiting for your response

kdiaz489, 2023-08-10:
Question in the previous comment was answered and supplemented with documents in a meeting that took place on August 9th at 10pm PST

darshanP-crest, 2023-08-11:
Here is the video demonstration of Transformation of EV Charger Timestamps into EV Charger Transactions testing on Postman and the corresponding database data entry

https://www.awesomescreenshot.com/video/19865690?key=c33dc25b2a93d48aa74293fca7d5eb89

darshanP-crest, 2023-08-14:
Questions from our side:

Q-1) Timestamp entries for EVSE error status codes should not be added to the database; only Zendesk ticket creation is required Right?

Q-2) How should we get transaction stop reason field?

darshanP-crest, 2023-08-16:
Here is the draw.io zip file and Image to demonstrate the flow of webhook.

[image]

[ALGO_FLOW.zip](https://github.com/kigt-inc/charge-cloud-backend-2023/files/12359331/ALGO_FLOW.zip)

darshanP-crest, 2023-08-17:
Here is the video demonstration of Creation sub transactions and testing on Postman.
https://www.awesomescreenshot.com/video/20005516?key=86aeb51ee74493d733e57fdf0a41829a
flow: 1 -> 2 -> 3 -> 2 -> 255

Here is the video demonstration of Creation sub transactions and testing on Postman.
https://www.awesomescreenshot.com/video/20005693?key=c5132171cbc54aece7237b82330e4ed4
flow: 1 -> 2 -> 3 -> 2 -> 254 -> 255