kigt-inc/charge-cloud-backend-2023 #17: creating algorithm for "EV charger_timestamp" to "EV charger station trans"

kigt-inc, 2023-08-10:


darshanP-crest, 2023-08-16:
Here is the draw.io zip file and Image to demonstrate the flow of webhook.

[image]

[ALGO_FLOW.zip](https://github.com/kigt-inc/charge-cloud-backend-2023/files/12359331/ALGO_FLOW.zip)

darshanP-crest, 2023-08-17:
Here is the video demonstration of Creation sub transactions and testing on Postman.
https://www.awesomescreenshot.com/video/20005516?key=86aeb51ee74493d733e57fdf0a41829a
flow: 1 -> 2 -> 3 -> 2 -> 255

Here is the video demonstration of Creation sub transactions and testing on Postman.
https://www.awesomescreenshot.com/video/20005693?key=c5132171cbc54aece7237b82330e4ed4
flow: 1 -> 2 -> 3 -> 2 -> 254 -> 255