kigt-inc/charge-cloud-backend-2023 #23: Handle error EVSE status code

kdiaz489, 2023-08-14:
Any EVSE errors that occur need to be reported to Zendesk via their API. The following should be sent in a POST request:

sender email, urgency level, subject, body

The login to our Zendesk account will be provided in the group chat on Teams if needed.

API Token Description: KIGT Charge Cloud Webhook 

ZENDESK_API_TOKEN = x147Ct2oBZ800yDaXePzCpYKhEq7o8uCtsa9N1M3 

https://developer.zendesk.com/api-reference/ticketing/tickets/tickets/

https://developer.zendesk.com/api-reference/ticketing/tickets/tickets/#create-ticket

manoj-k-crest, 2023-08-14:
here is the login for Zendesk, so you can see if the tickets were created from testing

 

Username: [jatomis@kigt.co]

Pass: Otra8v50n05ECvKlkckv

darshanP-crest, 2023-08-14:
Here is the video demonstration of Creation of Zendesk tickets according to EVSE error status codes and testing on Postman.
https://www.awesomescreenshot.com/video/19917323?key=fe44ed1f4341457b29f1435a7306d651

darshanP-crest, 2023-08-17:
Here is the video demonstration of Creation sub transactions and testing on Postman.
https://www.awesomescreenshot.com/video/20005516?key=86aeb51ee74493d733e57fdf0a41829a
flow: 1 -> 2 -> 3 -> 2 -> 255

Here is the video demonstration of Creation sub transactions and testing on Postman.
https://www.awesomescreenshot.com/video/20005693?key=c5132171cbc54aece7237b82330e4ed4
flow: 1 -> 2 -> 3 -> 2 -> 254 -> 255

Here is the video demonstration of Zendesk ticket body updation and testing on Postman.
https://www.awesomescreenshot.com/video/20006627?key=700f0c50387f43d921894e50a8b92d93