kigt-inc/charge-cloud-backend-2023 #24: Test Zendesk Ticket Creation via Postman

kdiaz489, 2023-08-14:
Please test out POST requests for Zendesk Ticket Creation via Postman

manoj-k-crest, 2023-08-14:
login url: https://kigt.zendesk.com/auth/v2/login/signin?return_to=https%3A%2F%2Fkigt.zendesk.com%2Fagent&theme=hc&locale=1&brand_id=360004197172&auth_origin=360004197172%2Cfalse%2Ctrue&role=agent

Username: [jatomis@kigt.co](mailto:jatomis@kigt.co) 

Pass: Otra8v50n05ECvKlkckv

manoj-k-crest, 2023-08-17:
From @kdiaz489 

cc @darshanP-crest 

Hi everyone! For the Zendesk error reporting, lets add more details in the body of the message sent to Zendesk. Lets add the current charger ID that is experiencing an error and the timestamp as well. Brandon Aparicio if there is anymore data that you feel would be useful, let us know.

Also the sub transaction is still not correct. In the example you showed in the video, there should be two transactions by the end of the session. One is the main transaction that records the time over all that the charger is being interacted with. The sub transaction is another transaction within the first one. 

It should be another entry in the transaction table that will only track when we enter status 3 (charing is enabled) and when charging is no longer provided to the car.

The video from yesterday was closer, but there were more transaction records created than was necessary.