kigt-inc/charge-cloud-backend-2023 #26: Create sub transactions

kdiaz489, 2023-08-14:
According to the architecture documents (V1, page 8) , there should be a sub transaction within the Charger Utilized type transaction called EV Charge Period. This sub transaction is compliant with OCPP. This will record actual charging time and energy use calculated. 

**Start of a EV Charge Period type transaction**
Charging is a change in state of the EV Charger. Based on the timing the CSMS can capture the
state of the EV Charger. Therefore when:

• 1 to a 3 – start a charging session related to the associated transaction for that EV
Charger.

- Set the EV Charger status (charge_station_status) will be set to “Charging”.
- Set the EV Charger connector status (connector_status) will be set to “Occupied”
- Set the EV Transaction status (transaction_status) will be set to “Charging”

• 2 to 3 – start a charging session related to the associated transaction for that EV
Charger.

- Set the EV Charger status (charge_station_status) will be set to “Charging”.
- Set the EV Charger connector status (connector_status) will be set to “Occupied” (this should already be the status)
- Set the EV Transaction status (transaction_status) will be set to “Charging”
- End of a charging session

Charging is a change is state of the EV Charger. Based on the timing the CSMS can capture the
state of the EV Charger. Therefore when:

- 3 to 2, or 3 to 254 to 2
- Set the EV Charger status (charge_station_status) will be set to “Connected”.
- Set the EV Charger connector status (connector_status) will be set to “Occupied”
- Set the EV Transaction status (transaction_status) will be set to “ended”, update
- the timestamp in the transaction to the timestamp of the change in status record.

The time stamp record will have the EVSE Energy Usage for the transaction,
divide it by 1000 for kwh and update the transaction record (KWH_Session).

darshanP-crest, 2023-08-14:
Here is the video demonstration of Creation sub transactions and testing on Postman.
https://www.awesomescreenshot.com/video/19924940?key=9ba16f34369c6cc9ac9da8df4412c3bb

darshanP-crest, 2023-08-17:
Here is the video demonstration of Creation sub transactions and testing on Postman.
https://www.awesomescreenshot.com/video/20005516?key=86aeb51ee74493d733e57fdf0a41829a
flow: 1 -> 2 -> 3 -> 2 -> 255

Here is the video demonstration of Creation sub transactions and testing on Postman.
https://www.awesomescreenshot.com/video/20005693?key=c5132171cbc54aece7237b82330e4ed4
flow: 1 -> 2 -> 3 -> 2 -> 254 -> 255