kigt-inc/charge-cloud-backend-2023 #27: Implement real time communication between backend and frontend

kdiaz489, 2023-08-14:
Real time updates between the MySql database and the React based front end is needed. We will need to know when chargers are available, occupied, etc. 

In the DB schema, there is a ChargeStation table, with a property called EVSE_Status_Code. I assume this record will provide all the info needed about Charge Stations we install. How can we keep these records up to date with the latest status code and other values pushed to the Webhook? 

After this is resolved, we can communicate these updates live to the front end via Socket io, or an equivalent method.

darshanP-crest, 2023-08-16:
Here is the video demonstration of Socket.io Implementation.
https://www.awesomescreenshot.com/video/19974523?key=0339a09b275a44a5a0c2a718c9193a4f

darshanP-crest, 2023-08-17:
Here is the video demonstration of Socket Charge Station Status and testing on Postman.
https://www.awesomescreenshot.com/video/20005424?key=9b326a683d2e830ad354639fa2aa65b2