kigt-inc/charge-cloud-backend-2023 #32: Develop backend operations necessary for RBAC (Role Based Access Control)

kdiaz489, 2023-08-17:
We will need to implement Role Based Access Control.
Right now, priority is creating a Client role and a Super Admin for KIGT employees/contractors to manage ALL chargers for ALL clients.

kdiaz489, 2023-08-17:
More information:

What is a Client (Site Owner)? 

Answer: Whoever/Entity that physically owns the KIGT charging station would be considered a client. Someone who receives the money  

Middle man between KIGT and the KIGT Charging stations 

The face of the operation 

Example, at UCR, they have 10 KIGT stations, but they own them, so they are the client 

They manage the chargers on their site  

What is a Merchant? 

KIGT is the merchant for now. This may be redefined soon. 

What is a Location? 

Answer: The location is the site of the chagers. For example the rancho cucamonga (club on 6th st) that hold 2 KIGT chargers or San Diego Convention Center that holds 10 KIGT chargers 

What is a Customer? 

Answer: Any Person that uses a charging station to charge their vehicle. They don’t own a charging station 

What is an Account in relation to a Customer? 

Answer: if customer is a KIGT member, they will have an account associated to them. Created by Mobile App right now? Confirm this.

manoj-k-crest, 2023-08-23:
[11:46 AM] manoj.kamothi (Guest)

What is the order of creations? First User wil be created Then Client and Then Locations?

[11:47 AM] Karen Diaz

yes that is correct ^ 

[11:50 AM] manoj.kamothi (Guest)

darshan.p (Guest) i think we need to modify schema

[11:50 AM] darshan.p (Guest)

manoj.kamothi (Guest)

darshan.p (Guest) i think we need to modify schema

Yes

[11:50 AM] manoj.kamothi (Guest)

I think we need to get rid of client id from users table and manage other relation with Users/Client/Locations.

[11:51 AM] Karen Diaz

yes i agree

[11:52 AM] manoj.kamothi (Guest)

Karen Diaz one concern is USER able to login into System?

[11:52 AM] Karen Diaz

User with Client role? yes 

[11:52 AM] Karen Diaz

User with Super Admin will be able to login as well 

[11:53 AM] manoj.kamothi (Guest)

Got it.

[11:54 AM] manoj.kamothi (Guest)

So Karen Diaz now we will add user_id as FK in Client table

[11:54 AM] manoj.kamothi (Guest)

we already have client_id in location table

[11:55 AM] Karen Diaz

So User with Super Admin role will login to an admin dashboard view where they can see all clients, locations and chargers

 

User with Client role will login in to a client dashboard view where they can only view graphs of reported data of their locations / chargers

[11:55 AM] Karen Diaz

sounds good

[11:55 AM] manoj.kamothi (Guest)

Sounds perfect

[11:55 AM] Karen Diaz

When you can, can you send an updated schema diagram

[11:57 AM] Karen Diaz

also, a User with a Client role will only ever have one Client associated with them. also confirmed this with brandon

 like 1

[11:57 AM] manoj.kamothi (Guest)

Yeah let us re think and update the schema

 like 1

darshanP-crest, 2023-08-23:
Here is the video demonstration of User -> client -> location -> chargeStation Creation flow with RBAC and testing on Postman.
https://www.awesomescreenshot.com/video/20164098?key=f3df31da1877ef4d1938f369a0157031

darshanP-crest, 2023-08-24:
Here is the video demonstration of Graphs response according to role and testing on Postman.
https://www.awesomescreenshot.com/video/20197414?key=c0ff90c197531ef973ddfbb27b055029