kigt-inc/charge-cloud-backend-2023 #33: Develop backend operations needed to graph information based on weekly/monthly/annual

kdiaz489, 2023-08-17:
once transactions are wrapped up, we will need to start thinking of how we can graph this data based on weekly/monthly/annually .

KIGT needs the following graphs on the UI:
 

Graph 1: Total amount of kilowats used in the charge (Energy Utilization) 
Graph 2: Total revenue per day (or time selected on graph)  
Graph 3: charges/transactions vs time 

We need the necessary backend code to make this possible, and possible pagination? So users can go through different weeks / months / years

Question: What is the most efficient way to calculate these values and pass it along to the front end react application?

darshanP-crest, 2023-08-21:
Here is the video demonstration of  Energy utilization and revenue generated Reports and testing on Postman.
https://www.awesomescreenshot.com/video/20097141?key=bcd63877c6891cb0f413b3f95408a7eb

Here is the video demonstration of  Charges and Transaction vs time Report and testing on Postman.
https://www.awesomescreenshot.com/video/20097603?key=49f10e9683f4f2a1bc3ba934f4946c4a

darshanP-crest, 2023-08-25:
Here is the video demonstration of third graph response according to role and testing on Postman.
https://www.awesomescreenshot.com/video/20223156?key=f1185cf0419a107cade16ed67f8147f8

darshanP-crest, 2023-08-25:
[image]