kigt-inc/charge-cloud-backend-2023 #46: Link Merchant to User account

kdiaz489, 2023-08-24:
This is similar to adding a Client to a User account.
According to our DB Schema, a relationship between a Merchant and Charge Stations can exist.
For the time being, KIGT is the Merchant for our two test sites. However, this may change in the future.
Please ask Brandon for more details on how Merchants work in relation to Clients (site owners) and Charge Stations. I believe it is who processes payments for the chargers and receives a portion of the money. Update this tickets with notes.

darshanP-crest, 2023-08-24:
Here is the video demonstration of Link Merchant to User account and testing on Postman.
https://www.awesomescreenshot.com/video/20196838?key=6b6896bf056bbc68254269ef78f287ad

darshanP-crest, 2023-08-24:
Here is the video demonstration of Graphs response according to role and testing on Postman.
https://www.awesomescreenshot.com/video/20197414?key=c0ff90c197531ef973ddfbb27b055029