kigt-inc/charge-cloud-backend-2023 #52: Add a "Test" Mode to the webhook

kdiaz489, 2023-09-05:
The KIGT team needs a way to test the web hook and only connect to ONE charger. At the moment, if we were to test out the webhook, it would connect to all chargers in the field which would not be efficient if we only want to test a specific charger.

Please create a new flag in the backend code for the webhook that lets it know if we are in test mode (boolean) or not. If we are in test mode, the tester needs to specify the charger id they want to connect to and allow incoming requests from that charger.

darshanP-crest, 2023-10-02:
@kdiaz489 Currently, we retrieve chargeStation data based on the serial number provided in the webhook's request body. This means that if we use the same serial number, we receive the identical chargeStation data each time.

darshanP-crest, 2023-10-04:
@kdiaz489 Any updates on this ticket?

kdiaz489, 2023-10-04:
@darshanP-crest my apologies, i dont understand the concern

darshanP-crest, 2023-10-04:
@kdiaz489 Currently, we obtain charge station data or establish a connection based on the serial number provided in the webbook's request body. Consequently, if we use the same serial number, we will consistently access the same charge station data. In the context of this ticket, while in test mode, we do not intend to connect to any additional charge stations, correct?

kdiaz489, 2023-10-04:
setting this task aside for now