kigt-inc/charge-cloud-backend-2023 #53: Fix adding location

kdiaz489, 2023-10-02:
Not able to add location to a user

[image]

darshanP-crest, 2023-10-02:
@kdiaz489 
This error happen because of the client with id 8 is does not exist on our database.

kdiaz489, 2023-10-02:
@darshanP-crest isnt this the user with id 8? 

[image]

darshanP-crest, 2023-10-02:
@kdiaz489 
No I am talking about client not the user!
According to our last discussion client owns the locations right?

kdiaz489, 2023-10-02:
@darshanP-crest Yes! My apologies i was using the user id. I will close this task

darshanP-crest, 2023-10-02:
@kdiaz489 It's Ok No problem!