kigt-inc/charge-cloud-backend-2023 #55: Add new event to track - Throttling

kdiaz489, 2023-10-04:
We want to monitor a new event - a Throttling event.
Throttling refers to when the current provided by the charger to the EV vehicle is lowered or increased to a certain amperage.
We should record the changes in amperage via an event timestamp and convert to a transaction once throttling is finished.

darshanP-crest, 2023-10-04:
@kdiaz489 If the timestamp of any EV charger contains a Max Current value less than 28, then the KWh for both the overall transaction and the sub-transactions will be calculated as follows: (duration (in minutes) / 60) * EVSE Max Current for that specific charging session. Is that correct?

kdiaz489, 2023-10-04:
@darshanP-crest Lets add to both for now, then we will adjust after more research.

darshanP-crest, 2023-10-04:
@kdiaz489
For instance, let's consider a charge session that goes like this: 1 -> 2 -> 3 -> 2 -> 255. During this session, we observe that the MAX Current is 20 at the timestamp when the EVSE status code is "1." After that, we see the MAX Current as 28 when the EVSE status code is "2." This same pattern occurs between the transitions from "3" to "2" in the EVSE status code.

In this scenario, we need to record a total of 4 transactions in the ev_charge_station_trans table:

1) The overall transaction
2) The sub-transaction
3) The first throttling event during the transition from "1" to "2"
4) The second throttling event during the transition from "3" to "2"

Is that Right?

kdiaz489, 2023-10-05:
@darshanP-crest 

According to my conversation with @brandon the conditions for throttling to start being recorded are the following: 
status should be 3, and max current should turn into a number less than 28. 
This is scheduled/initiated by the user. 

Throttling is an event that normally occurs when a vehicle is charging. Status code 3 signals that the vehicle is receiving power, so we want to record if the max current changes during that period. From 1 -> 2 , the vehicle is not actually receiving power, so we dont need to record that.

darshanP-crest, 2023-10-05:
@kdiaz489
Yes, you're absolutely right. I was thinking the same thing yesterday, that throttling should occur when a vehicle is charging. So, So, I am implementing it in this manner.

kdiaz489, 2023-10-05:
additional info from brandon

Throttling events should only be registered during state "3".
Initiate a sub-transaction for the throttling event when the “MAX CURRENT” deviates from the default value of "28".
End the sub-transaction once the “MAX CURRENT” returns to "28" or when the charging session ends.
Record two separate power values:
a) Power consumption during throttling.
b) Power consumption during regular charging.
Please note: Throttling will be influenced by user input.
Ensure that the charging event transaction remains active if the session is neither stopped nor paused.

darshanP-crest, 2023-10-05:
@kdiaz489 Noted!

darshanP-crest, 2023-10-06:
Here is the video demonstration of Add new event to track - Throttling & testing on the Postman
https://www.awesomescreenshot.com/video/21338406?key=e35be9436676dc5b7683090162eee2e7

darshanP-crest, 2023-10-09:
@kdiaz489 

[image]

To fulfill this objective, could you specify the field and table that require modification?