kigt-inc/charge-cloud-backend-2023 #60: Charging Event Trigger

kigt-inc, 2023-10-09:
Implement a database trigger that records the exact time a user begins a the process of starting a charge event (i.e., when the button "Start" on the Home Screen is pressed).

darshanP-crest, 2023-10-11:
@kigt-inc @kdiaz489 
Which field and table need to be modified to achieve this objective?

darshanP-crest, 2023-10-11:
Here is the video demonstration of Screen interaction trigger & testing on the Postman
https://www.awesomescreenshot.com/video/21491723?key=d0de60ac1a9027f5f2a2903012b380bf

Please note that in the video, the 'ev_charge_station_trans' table is used, and it involves three transactions:
1. The first transaction is the main transaction.
2. The second transaction is a sub-transaction used to record the time a customer spends on the interface.
3. The third transaction is a sub-transaction used to record the charging event.

darshanP-crest, 2023-10-11:
Here is the video demonstration of Screen interaction trigger with throttling event & testing on the Postman
https://www.awesomescreenshot.com/video/21491840?key=a14ef4b0df02f473ed5778e2aae5cd12

Please note that in the video, the 'ev_charge_station_trans' table is used, and it involves four transactions:
1. The first transaction is the main transaction.
2. The second transaction is a sub-transaction used to record the time a customer spends on the interface.
3. The third transaction is a sub-transaction used to record the charging event.
4. The fourth transaction is a sub-transaction used to record the throttling event.